ArduPilot/ArduRemoteID #21: Add ability to receive RemoteId from other drones

rmackay9, 2022-08-18:
It could be useful for various reasons to be able to receive the location of other drones that are publishing using a remote id.  Some examples are:

- warning the pilot of other drones in the area
- object avoidance of other drones
- automatic monitoring/filming of drones in the area

Junnicutt, 2022-08-27:
Does any of the esp32 c3 or s3 firmware currently used for opendroneid support receiving any opendroneid transmissions? Or is there a different firmware that would do this? I know the goal is something that can work easily on most cell phones but having esp32s able to receive could have lots of uses like vehicle to vehicle or long term monitoring of an area.

friissoren, 2022-08-29:
For the older ESP32 HW (non C3/S3), there is some receiver code available here: https://github.com/sxjack/uav_electronic_ids/tree/main/id_scanner

I haven't tested this. I don't know whether it can run both transmit and receive simultaneously.