ArduPilot/ArduRemoteID #32: ArduRemoteID Module - UA Serial Number, Serial Number Type and Vehicle Type Burn In

sborenstein-ift, 2022-08-26:
In order to comply with MOC 7.5.2, the serial number needs to be baked into the module. It is recommended that this be done for all three fields required for BASIC_ID. This includes: 

- UAS ID: 20 characters 
- UA Type: MAV_ODID_UA_TYPE (byte)
- ID Type: MAV_ODID_TYPE

One recommendation is to burn this in on the mfg floor at firmware burn-in time. Other options include a one-time write capability after firmware is loaded. Other options may exist, however there seems to be general agreement that:

1) This information should be baked into the module, not the autopilot
2) The contents of the 20 character ID number should start with the 4 character ICAO requirement, but otherwise be vendor specific.

BluemarkInnovations, 2022-08-29:
The MOC 7.5.2 does not mandate that it is baked into the module. It requires that:

_The Remote ID system shall protect ... message elements from being altered through the end-user interface(s) of the system by masking the items from user input._

_NOTE 7—End-user interfaces include the following: External Buttons, screens, add-on devices (such as cell phones/apps), external wiring interfaces + supporting software._

In my opinion, the Remote ID system does not mean the module only, but instead the whole auto pilot system. In the current implementation of the ArduRemoteID firmware, it is a dumb device and has no end-user interface. There is only internal wiring from the auto pilot system to the module.

> Other options may exist, however there seems to be general agreement that:
> 1. This information should be baked into the module, not the autopilot

Do you have references that support this?

sborenstein-ift, 2022-08-29:
@BluemarkInnovations The MOC 7.5.2 mandates that the message elements be masked from the user, and only allow for user input of the MODE of ID (Session or Serial) and Emergency Status. Thus, in order to meet compliance, the serial number must be integrated into the solution. The UA Type is also required in the message according to F3411, so this should be baked in, as well. The classification and classification type are region dependent.

sborenstein-ift, 2022-08-29:
BTW - keep in mind that we're using this issue repository as a location to capture ALL related ArduPilot and ArduRemoteID related issues, as discussed on Discord. Not just ArduRemoteID module firmware related.

BluemarkInnovations, 2022-08-29:
I think it basically boils down to what the definition of a Remote ID system is. In your opinion it is the module only, in my opinion, it is the auto pilot system + module.

> BTW - keep in mind that we're using this issue repository as a location to capture ALL related ArduPilot and ArduRemoteID related issues, as discussed on Discord. Not just ArduRemoteID module firmware related.

Yes, I know. But you mention in the start post the word "module". So I relate this to the ArduRemoteID firmware. Let's wait for others to comment and have their view.

sborenstein-ift, 2022-08-29:
Yes, I did. That is because immediately prior to posting this, @tridge and I discussed that the RID module made the most sense to store this data, as he AP parameters weren't type appropriate for it.

BluemarkInnovations, 2022-08-29:
Then I should talk to @tridge as I didn't do that since you started this issue. So we avoid discussion loops.

friissoren, 2022-08-30:
I agree that the serial number should be (securely) stored in the RID transmitter module (or in the flight controller HW, if those two are tightly integrated). I went through both ASTM documents again and surprisingly there weren't an explicit requirement in those to have it stored in a secure area (at least I couldn't find it right now) but the ASD-STAN document for Europe explicitly states:

_The Direct Remote Identification system shall reduce the ability of tampering with the functionality of the direct remote identification system.
The Direct Remote Identification system is the on-board feature in the UA/Add-on that is formatting and transmitting over the air the DRI information to a compatible receiver mobile device.
**The UA and the Add-on come with a unique serial number; this number is loaded at factory level and shall not be modified anymore afterwards. The protection of the unique serial number of the UA/Add-on shall be done by design. The serial number shall be stored in a secure memory area.**
The design of the DRI shall not allow the user to modify the Geolocation information part of the DRI message, as defined in Clause 4.3 above.
The conformity to those requirements shall be proven per the OEM’s design documentation._

Keep in mind also, that the [Declaration of Compliance](https://uasdoc.faa.gov/login) document needs to have a list of all the serial numbers that have been manufactured and being sold. Chapter 5.9 of this [document](https://www.faa.gov/documentLibrary/media/Advisory_Circular/AC_89-2.pdf).

BluemarkInnovations, 2022-10-02:
@sborenstein-ift I think this issue is solved in in version 1.9+. The LOCK_LEVEL=2  feature makes it baked into the module. Only stakeholders with the correct private key can change those values.

>The contents of the 20 character ID number should start with the 4 character ICAO requirement, but otherwise be vendor specific.

It is up to the stakeholder with the private key to configure a correct ID number that meets the local regulation, such as the 4-char ICAO requirement.

tridge, 2022-11-09:
this is solved with the LOCK_LEVEL parameter, which should be set by the vendor

closed 2022-11-09 by tridge