ArduPilot/ArduRemoteID #33: OpenDroneID - Need to add ENUM types for MAV_ODID_ID_TYPE

sborenstein-ift, 2022-08-26:
This will be a MAVLink issue. 

Currently, MAV_ODID_ID_TYPE only contains CAA. Need to add other options.

friissoren, 2022-08-26:
Sorry, could you elaborate?

MAVLink has the available [enum values](https://mavlink.io/en/messages/common.html#MAV_ODID_ID_TYPE)

And those are there in the [ArduPilot variant](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L3296) also.

In the core-c library, the corresponding ones [are here](https://github.com/opendroneid/opendroneid-core-c/blob/master/libopendroneid/opendroneid.h#L123)

sborenstein-ift, 2022-08-29:
Whoops... this is supposed to be fore MAV_ODID_OPERATOR_ID_TYPE.

friissoren, 2022-08-30:
Sorry, I still don't understand what is missing. 

MAVLink has it defined [here](https://mavlink.io/en/messages/common.html#MAV_ODID_OPERATOR_ID_TYPE). And core-c [here](https://github.com/opendroneid/opendroneid-core-c/blob/master/libopendroneid/opendroneid.h#L244). ArduRemoteID using it [here](https://github.com/ArduPilot/ArduRemoteID/blob/master/RemoteIDModule/RemoteIDModule.ino#L152).

The standards do not have any other values defined than zero for this enum field. Unless you want to spell out the "Reserved" and "Available for private use" ranges.

tridge, 2022-11-09:
this seems to be fine

closed 2022-11-09 by tridge