ArduPilot/ArduRemoteID #54: Operator Location doesn't fail on CAN

sborenstein-ift, 2022-09-16:
This may be a bug on the ArduPilot OpenDroneID library. When the operator location feed is turned off, the RID module does not detect this. Likely due to the AP module not failing and continuing to send stale data.

BluemarkInnovations, 2022-10-04:
Maybe this is a feature. 

In PR https://github.com/ArduPilot/ArduRemoteID/pull/79 I added a more sophisticated check in which the  arm_status_check function will also fail if the flight controller keeps sending the same system/location packets.

In general it is true that the RID module will keep broadcasting the last received information. Also, if the operator location feed is turned off. The developer choice is a) broadcast the last know information b) make the information invalid. 

For instance, in case of the location messages, it is more useful to broadcast the last know location, compared to make it invalid. Similar to the operator location. It is better to broadcast the last know information then no information at all.

tridge, 2022-11-09:
fixed

PR #90 (2022-11-09): fixed web server status web operator location is lost
fixes #54
commit c23905a1 (2022-11-09): fixed web server status web operator location is lost

fixes #54
closed 2022-11-09 by tridge