ArduPilot/ArduRemoteID #93: Troubleshooting Guidance

nichogenius, 2022-11-27:
I am attempting to get this working on an ESP32-C3 dev board and I am running into a dead end.  I could use some troubleshooting advice.

I believe I have successfully flashed the ESP32-C3 board, but I can't see any indication that it is working.  I have a constant red LED indicating power as the only sign of life.  No wifi networks visible on my phone, no bluetooth devices visible on my phone, no indication that mavlink is being sent from my flight controller, and my drone scanner apps can't see anything.

What tests can I run to verify my ESP32-C3 ArduRemoteID flash was successful?

Ideas for what might be wrong:
1. I missed some part of the setup procedure in Mission Planner (I am very new to DIY drones and Mission Planner)
2. Insufficient power from my USB port to power all devices
3. Using wrong pins on the ESP32-C3 for TX/RX . The readme has pin 2 as RX and pin 3 as TX which is backwards from the board labels.  I tried both just in case.
4.  Bad flash.  I tried two different flashing utilities, neither of which worked.  I finally got it to flash using esptool.py from Windows Command Line.
5. Bad wiring between the flight controller and the ESP32-C3.  I am new to soldering.

My Setup
-----
Breadboard:
  - F405-WMN:
     - ArduPlane V4.3.1
     - Powered via USB to Flight Controller (No Battery)
     - Active Mavlink Connection to Mission Planner via USB
  - GPS/Compass (Calibrated and Working):
     - Power from Flight Controller 5v
  - ESP32-C3 Dev Kit:
     - Flashed ArduRemoteID release: v1.11, file: ArduRemoteID_ESP32C3_DEV_OTA.bin
     - Power from Flight Controller 5v
     - Connected to UART1 TX/RX on flight controller

Desktop PC (Windows 10):
  - Mission Planner v1.3.77.1 (build 1.3.8349.13456)
     - Telem1 set to use Mavlink2
     - GCS GPS from Android Phone via bluetooth (Connected and Working)
  - USB Bluetooth Dongle (to receive GPS from Android Phone)

Android Phone:
  - GPS2Bluetooth
  - Drone Scanner (Application to test the transponder)
  - OpenDroneID (Application to test the transponder)
-----

BluemarkInnovations, 2022-11-27:
I wrote most of the comment before I saw your update 2. Good that it works now! You can only use OTA (Over The Air) firmware files to update via the web interface. For the initial flashing you need to use the normal firmware file.

- If you flash it successfully, the ESP32-C3 will emit BT4/BT5 packets even if there is no connection with the flight controller. Hence, you could connect the ESP32-C3 to a USB charger only and receive data with the Android app. Of course, most of the payload will be empty.
- Do you have a ESP32-C3 with an internal or external antenna (u.FL connector)? If the latter, you need to connect an antenna.
- Both the Drone Scanner and OpenDroneID app **need** location permissions for correct BT reception.
- Typically, the ArduRemoteID firmware will output debug messages on the serial console. Do you see any serial output if you connect a serial console program e.g. putty? I think default baud rate is 115200.
- Once you start Mission Planner and configured the OpenDroneID settings: https://ardupilot.org/dev/docs/opendroneid.html#opendroneid like DID_ENABLE, etc, an ODID_TXRX device should appear if you connect to AP (see the screen shot below).
- A USB 2.0 port may not deliver enough power for your whole setup. Use a USB3.0 port.

[image]

nichogenius, 2022-11-27:
Thanks for all of the wonderful advice!  I think I'm on the last issue:

[image]

BluemarkInnovations, 2022-11-27:
RemoteID needs two GPS receivers. One for the drone location and the other one for the operator location. So the issue is the GPS receiver connected to the flight controller. Did you configure that correctly? (Mission Planner configuration) If you have set it up correctly, you should see the drone location in Mission Planner regardless of the RID configuration. It could also be that you connected a GPS receiver/chip to the flight controller that is not supported. For instance u-blox 8/9 series are supported. FYI there is also the #opendroneid channel in the ArduPilot discord server for support.

nichogenius, 2022-11-27:
@BluemarkInnovations Thank you! Hardware compatibility is my top suspect right now.  I'll close this and move any further discussion over to Discord.  Thanks again!

closed 2022-11-27 by nichogenius

nichogenius, 2022-11-27:
The problem with the missing location message was resolved by flashing my flight controller with a firmware build that had ODID enabled.  As ArduPlane is not currently shipped with ODID enabled, I had to compile it myself.  Leaving this here for anyone that finds this.