ArduPilot/ArduRemoteID #97: Unable to connect to DroneCanGUI Tool or Ardupilot

dairoca90, 2023-01-11:
I was trying to test the ArduRemoteID with the recommend hardware the ESP32-S3 dev board: https://au.mouser.com/ProductDetail/356-ESP32S3DEVKTM1N8, but I cannot se the device in either DroneCanGUI Tool (here I'm using the usb and uart connection ) or Ardupilot software (using can bus explained in the Ardupilot documentation) but in either case I'm getting the configuration data of the DroneCanDevice, don't know if i have to send opendroneid messages in order to activate the device or em I missing something, the device RGB just turn red even one I connected to Pixhawk can bus with opendroneid firmware enabled.

[image]

BluemarkInnovations, 2023-01-11:
I think the basic answer is that the ArduRemoteID device can interface via UART (MAVLink) or CAN bus (DroneCAN) to the flight controller. The ESP32-S3 has only an UART interface. If you want to use the CAN bus you need to add an external CAN transceiver chip. You connected the ESP32-S3 using UART, so DroneCanGUI Tool does not work.

Mission Planner can configure ArduRemoteID devices via MAVLink (only non-stirng values). See Chapter 4: https://download.bluemark.io/db200.pdf

dairoca90, 2023-01-11:
I already tried to connect using an external SN65HVD230 CAN Bus Transceiver but I can't get any information, that's why I try to use DroneCan GUI with the usb and serial connection, in the bluemark device documentation it says that it can be configured using DroneCan GUI without using Mission Planner but it doesnt say how can be connected to de computer to read the paramaters.

BluemarkInnovations, 2023-01-12:
So you need to connect the RID transmitter to the Pixhawk. And the Pixhawk to your computer using USB. There is a YouTube video:
https://www.youtube.com/watch?v=Az8v4Kx4hS0&si=EnSIkaIECMiOmarE&t=26

In the [board_config.h](https://github.com/ArduPilot/ArduRemoteID/blob/master/RemoteIDModule/board_config.h) you can see which pins are used for each connection (MAVLink/DroneCAN). For your device it is the BOARD_ESP32S3_DEV.

dairoca90, 2023-01-12:
I'm switching to mavlink, I don't know if my CAN transceiver is the problem or pixhawk settings, but I use a serial connection and that made it work, thanks for the help, as ArduRemoteID is working im closing the issue.

closed 2023-01-12 by dairoca90