ArduPilot/ArduRemoteID #112: Some questions about ArduRemoteID on FMUV3

jianjianliu0918, 2023-04-25:
According to the tutorial on youtube, I tried to compile and burn the firmware with ODID support on FMUV3. 
My device is FMUV3, ardupilot version is 4.3.1, MP version is 1.3.80.
But after entering 'flashbootloader' using MAVProxy terminal, the device became bricked. After unbricking, I recompiled and burned again, this time I skipped the 'flashbootloader' step. But in the UAVCAN interface on MP, I didn't get the same result as on the video, see
[image]
Also, the MP didn't find ODID_TXRX device and the messages column shows as
[image]

Do I need to use a newer device? Like the pixhawk6c?

BluemarkInnovations, 2023-06-06:
I think you need to add a RemoteID module to your setup. Did you add that to your FMUV3 board?

jianjianliu0918, 2023-06-15:
> I think you need to add a RemoteID module to your setup. Did you add that to your FMUV3 board?

Hello,thank you for your reply.
I have added the RemoteID module as the tutorial, but Ardupilot's bootloader can not be updated, probably because there is not enough flash memory. 
I've tried a few things with the Pixhawk6C device, but it hasn't worked yet.