ArduPilot/ArduRemoteID #114: Can't compile on Ubuntu 22.04

shalomc-airo, 2023-05-17:
After following the instructions in BUILDING.md, I seethe following compilation error:

`Building ESP32S3_DEV on esp32s3
In file included from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink_msgs.h:18,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/transport.h:6,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink.h:5,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/RemoteIDModule.ino:15:
/media/shalomc/home/Arduino/libraries/mavlink2/mavlink2.h:1:10: fatal error: generated/all/version.h: No such file or directory
 #include "generated/all/version.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`

**Host**: Ubuntu 22.04
**Python**: 3.10.6

**Make output**:
`
➜  RemoteIDModule git:(master) make setup 
Installing ESP32 support
../bin/arduino-cli core update-index --config-file arduino-cli.yaml
Downloading index: package_index.tar.bz2 downloaded                                                                                                   
Downloading index: package_esp32_index.json downloaded                                                                                                
../bin/arduino-cli core install esp32:esp32@2.0.3
Platform esp32:esp32@2.0.3 already installed
../bin/arduino-cli lib install "Adafruit NeoPixel"
Already installed Adafruit NeoPixel@1.10.5
➜  RemoteIDModule git:(master) make
Generating mavlink2 headers
Validating modules/mavlink/message_definitions/v1.0/all.xml
Parsing modules/mavlink/message_definitions/v1.0/all.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ASLUAV.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ASLUAV.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/common.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/common.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/development.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/development.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/icarous.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/icarous.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/minimal.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/minimal.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/python_array_test.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/python_array_test.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/standard.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/standard.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/test.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/test.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ualberta.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/ualberta.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/uAvionix.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/uAvionix.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/loweheiser.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/loweheiser.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/storm32.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/storm32.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/AVSSUAS.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/AVSSUAS.xml
Validating /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/cubepilot.xml
Parsing /media/shalomc/home/workspace/opensource/arduremoteid/modules/mavlink/message_definitions/v1.0/cubepilot.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Merged enum MAV_CMD
Enum STORAGE_TYPE in STORAGE_INFORMATION.type does not exist
Generating DroneCAN headers for libcanard
expanding uavcan.Timestamp
expanding uavcan.CoarseOrientation
[... 97 lines cut ...]
Embedding file web/images/logo.jpg:web/images/logo.jpg
Embedding file web/index.html:web/index.html
Embedding file web/js/jquery.min.js:web/js/jquery.min.js
Embedding file web/js/tools.js:web/js/tools.js
Embedding file web/styles/main.css:web/styles/main.css
Building ESP32S3_DEV on esp32s3
In file included from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink_msgs.h:18,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/transport.h:6,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink.h:5,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/RemoteIDModule.ino:15:
/media/shalomc/home/Arduino/libraries/mavlink2/mavlink2.h:1:10: fatal error: generated/all/version.h: No such file or directory
 #include "generated/all/version.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Used library   Version Path                                                                                           
libopendroneid         /media/shalomc/home/workspace/opensource/arduremoteid/modules/opendroneid-core-c/libopendroneid
mavlink2       1.0     /media/shalomc/home/workspace/opensource/arduremoteid/libraries/mavlink2                       

Used platform Version Path                                                              
esp32:esp32   2.0.3   /media/shalomc/home/.arduino15/packages/esp32/hardware/esp32/2.0.3

Error during build: exit status 1
make: *** [Makefile:61: ArduRemoteID-ESP32S3_DEV.bin] Error 1
`

shalomc-airo, 2023-05-18:
Looks like the problems is that mavgen was failing, this was with latest head (Hash 9806a3ba7).
Changing to branch ardupilot-stable-4.1 allowed the compilation to continue, but it still failed with errors like 
```
In file included from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink.h:5,
                 from /media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/mavlink.cpp:5:
/media/shalomc/home/workspace/opensource/arduremoteid/RemoteIDModule/transport.h:18:11: error: 'mavlink_open_drone_id_location_t' does not name a type; did you mean 'mavlink_set_home_position_t'?
     const mavlink_open_drone_id_location_t &get_location(void) const {
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           mavlink_set_home_position_t
```

shalomc-airo, 2023-05-18:
This seems to be a problem with versions of the submodules. There is Ardupilot and under that, Mavlink.
Checking out ardupilot-stable-4.1 and Mavlink master seems to have solved the compilation problem.

My experience with these things is that you have to limit yourself to specific commits of each of the submodules and enforce this in the setup scripts. This is what I recommend for you.

BluemarkInnovations, 2023-05-23:
Thanks for reporting. Typically we stick to a specific commit of the submodule. However, the master branch has been moved recently to a new MAVLink version due to this PR: https://github.com/ArduPilot/ArduRemoteID/pull/95 

I guess that caused the compile errors (and was tested insufficiently). The automatic build of my latest PR fails with similar compile errors now. I will discuss @tridge to solve this issue.

kc2rxo, 2023-06-05:
> Checking out ardupilot-stable-4.1 and Mavlink master seems to have solved the compilation problem.

Setting `modules/mavlink` to `ardupilot-stable-4.1` and `modules/mavlink/pymavlink` to `master` did not seem to fix the compilation problems (which are the same) for me.

Is there a specific commit that did work for you 3 weeks ago?

BluemarkInnovations, 2023-06-05:
Can you use commit: 5b4e0414e73f0bef1328c60c953a58f895f9db14 ?

It is the commit of the 1.11 release: https://github.com/ArduPilot/ArduRemoteID/commit/5b4e0414e73f0bef1328c60c953a58f895f9db14

Important also execute the steps in section 2: get code of [BUILDING.md](https://github.com/ArduPilot/ArduRemoteID/blob/master/BUILDING.md#step2-get-code) again. 

I moved from Ubuntu to a Debian OS build environment (as virtual machine, to keep the build environment clean) where 1.11 compiles without issues.

BluemarkInnovations, 2023-06-05:
So I gave the wrong answer. I just downloaded a fresh ArduRemoteID project and compiled it without issues under Debian: bookworm/sid. The issue seems to be Ubuntu related(?)
```
roel@arduremoteid:~$ cd /tmp/
roel@arduremoteid:/tmp$ git clone https://github.com/ardupilot/arduremoteid
Cloning into 'arduremoteid'...
remote: Enumerating objects: 1009, done.
remote: Counting objects: 100% (361/361), done.
remote: Compressing objects: 100% (165/165), done.
remote: Total 1009 (delta 229), reused 205 (delta 196), pack-reused 648
Receiving objects: 100% (1009/1009), 2.24 MiB | 1.96 MiB/s, done.
Resolving deltas: 100% (651/651), done.
roel@arduremoteid:/tmp$ cd arduremoteid/
roel@arduremoteid:/tmp/arduremoteid$ git submodule init
Submodule 'modules/DSDL' (https://github.com/dronecan/DSDL.git) registered for path 'modules/DSDL'
Submodule 'modules/dronecan_dsdlc' (https://github.com/dronecan/dronecan_dsdlc.git) registered for path 'modules/dronecan_dsdlc'
Submodule 'modules/libcanard' (https://github.com/dronecan/libcanard.git) registered for path 'modules/libcanard'
Submodule 'modules/mavlink' (https://github.com/ArduPilot/mavlink.git) registered for path 'modules/mavlink'
Submodule 'modules/opendroneid-core-c' (https://github.com/opendroneid/opendroneid-core-c.git) registered for path 'modules/opendroneid-core-c'
roel@arduremoteid:/tmp/arduremoteid$ git submodule update --recursive
Cloning into '/tmp/arduremoteid/modules/DSDL'...
Cloning into '/tmp/arduremoteid/modules/dronecan_dsdlc'...
Cloning into '/tmp/arduremoteid/modules/libcanard'...
Cloning into '/tmp/arduremoteid/modules/mavlink'...
Cloning into '/tmp/arduremoteid/modules/opendroneid-core-c'...
Submodule path 'modules/DSDL': checked out '04e821222bd5407898861683ad0738c28b8fcaaf'
Submodule path 'modules/dronecan_dsdlc': checked out '6a252d2e33e88144abda7e28e4ded49465dd9912'
Submodule path 'modules/libcanard': checked out 'f646cdd9345c79b2d144056dcce0fd342a3da12f'
Submodule path 'modules/mavlink': checked out '806a3ba7499da677ded35226d6ff88e5a100c4c2'
Submodule path 'modules/opendroneid-core-c': checked out 'ebcff1e44fa549c09d847ac93deffceb8983aec2'
roel@arduremoteid:/tmp/arduremoteid$ ./scripts/install_build_env.sh
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: empy in /home/roel/.local/lib/python3.10/site-packages (3.3.4)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pymavlink in /home/roel/.local/lib/python3.10/site-packages (2.4.37)
Requirement already satisfied: lxml in /home/roel/.local/lib/python3.10/site-packages (from pymavlink) (4.9.2)
Requirement already satisfied: future in /home/roel/.local/lib/python3.10/site-packages (from pymavlink) (0.18.3)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: dronecan in /home/roel/.local/lib/python3.10/site-packages (1.0.17)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (3.5)
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pexpect in /home/roel/.local/lib/python3.10/site-packages (4.8.0)
Requirement already satisfied: ptyprocess>=0.5 in /home/roel/.local/lib/python3.10/site-packages (from pexpect) (0.7.0)
--2023-06-05 20:07:17--  https://downloads.arduino.cc/arduino-cli/arduino-cli_0.27.1_Linux_64bit.tar.gz
Resolving downloads.arduino.cc (downloads.arduino.cc)... 104.18.12.241, 104.18.13.241, 2606:4700::6812:cf1, ...
Connecting to downloads.arduino.cc (downloads.arduino.cc)|104.18.12.241|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13011374 (12M) [application/octet-stream]
Saving to: ‘arduino-cli_0.27.1_Linux_64bit.tar.gz’

arduino-cli_0.27.1_Linux_64bit.tar.gz  100%[============================================================================>]  12.41M  2.62MB/s    in 7.9s    

2023-06-05 20:07:26 (1.57 MB/s) - ‘arduino-cli_0.27.1_Linux_64bit.tar.gz’ saved [13011374/13011374]

arduino-cli
LICENSE.txt
roel@arduremoteid:/tmp/arduremoteid$ ./scripts/regen_headers.sh
Generating mavlink2 headers
Validating modules/mavlink/message_definitions/v1.0/all.xml
Parsing modules/mavlink/message_definitions/v1.0/all.xml
Validating /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Parsing /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Validating /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/ASLUAV.xml
Parsing /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/ASLUAV.xml
Validating /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/common.xml
Parsing /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/common.xml
Validating /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/development.xml
Parsing /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/development.xml
Validating /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/icarous.xml
Parsing /tmp/arduremoteid/modules/mavlink/message_definitions/v1.0/icarous.xml
[... 424 lines cut ...]
PyGObject             3.42.1
PyJWT                 2.3.0
pymavlink             2.4.37
pymonocypher          3.1.3.1
pyparsing             2.4.7
pyrsistent            0.18.1
pyserial              3.5
python-apt            2.3.0+ubuntu2.1
pytz                  2022.1
PyYAML                5.4.1
requests              2.25.1
SecretStorage         3.3.1
setuptools            59.6.0
six                   1.16.0
ssh-import-id         5.11
systemd-python        234
ubuntu-drivers-common 0.0.0
unattended-upgrades   0.1
urllib3               1.26.5
wadllib               1.3.6
wheel                 0.37.1
xkit                  0.0.0
zipp                  1.0.0

```

kc2rxo, 2023-06-05:
Double checked a few things. First I am running Ubuntu 20.04.6. Second my `pymavlink` was at `2.4.18` which I force upgraded via pip to `2.4.39`. I am running Python 3.8.10.

This upgrade did nothing to fix the issue. I going to try and use a VM with a super clean install and see what happens.

kc2rxo, 2023-06-05:
Both VMs I setup (Ubuntu 22.04.2 LTS and Debian 11.7) fail with a clean repo pull and following `BUILDING.md`.

They both fail with original error reported by @shalomc-airo.

I tried a lot of combinations most of which I failed to jot down. What did work on Debian however was this:

~~~bash
git clone https://github.com/ardupilot/arduremoteid
cd ArduRemoteID
git submodule init
git submodule update --recursive
cd modules/mavlink
git checkout master
cd ../..
./scripts/install_build_env.sh
./scripts/regen_headers.sh
./scripts/add_libraries.sh
cd RemoteIDModule
make setup
make esp32c3dev
~~~

Ubuntu stills fails complaining about missing `generated/all/version.h` using the same steps.

During testing I would blow away the repo completely and start from scratch to be sure everything I could was cleaned out.

shalomc-airo, 2023-06-06:
The versions that compiled for me on Ubuntu 22.04:
arduremoteid: ddafa38
mavlink: 24e31ed3
libcanard: f646cdd
dronecan_dsdlc: 6a252d2
DSDL: 04e8212

kc2rxo, 2023-06-06:
@shalomc-airo your latest works for me as well on Ubuntu 20.04.

Just to complete the list of commits:
arduremoteid: ddafa38
mavlink: 24e31ed3 / pymavlink: f41a355
libcanard: f646cdd
dronecan_dsdlc: 6a252d2
DSDL: 04e8212
opendroneid-c-core: ebcff1e4

kc2rxo, 2023-06-06:
Ran a clean build on Ubuntu 20.04, all works with the following commands:

~~~bash
git clone https://github.com/ArduPilot/ArduRemoteID.git
cd ArduRemoteID
git submodule init
git submodule update --recursive
git checkout ddafa387
cd modules/mavlink
git checkout 24e31ed3
cd ../..
./scripts/install_build_env.sh
./scripts/regen_headers.sh
./scripts/add_libraries.sh
cd RemoteIDModule
make setup
make esp32c3dev
~~~

Commits again for everything:
arduremoteid: ddafa38
modules/mavlink: 24e31ed3
modules/libcanard: f646cdd
modules/dronecan_dsdlc: 6a252d2
modules/DSDL: 04e8212
modules/opendroneid-c-core: ebcff1e4

tridge, 2023-06-15:
I think the issue is just the submodules were not updated. You can re-run
```
 git submodule update --recursive --force --init
```
when you have done an update of this project and that will update your submodules to the right version

BluemarkInnovations, 2023-06-16:
With the latest master (fc606790181b97f16963f474f459118508bbeea6), I was able to build ArduRemoteID without errors.