ArduPilot/ArduRemoteID #138: Operator id issue

ryanjAA, 2024-01-25:
In the 1.14 firmware (likely previous ones as well) it requires an operator id otherwise you get the op_id arm error. Since only required in EU jurisdictions some type of mechanism to disable is needed for US based flights. @BluemarkInnovations

BluemarkInnovations, 2024-01-25:
@ryanjAA There are two issues at play. PX4 decided (for now) not to implement the OperatorID message. ArduPilot has implemented it and streams empty OperatorID data in the USA. Solution one is that PX4 implements this message type also. In https://github.com/ArduPilot/ArduRemoteID/pull/101 (solution 2) I already suggested to have an region parameter, so you can do checks per region. However, that has not been merged yet in to a stable release.

ryanjAA, 2024-01-26:
What about https://github.com/PX4/PX4-Autopilot/pull/21647 even with this it still is an issue because while op id is transmitted if nothing is populated than it is red on the QGC side (easy to fix) but then on the rid side if something is typed in
we see invalid data said when picking up the module from phone/tablet. I pinged on discord as well. Working on getting this finished on px4   

I’m wondering if just omitting op id for FAA based ops. New firmware branch on the rid module side.

BluemarkInnovations, 2024-01-26:
Too busy, so I skipped Discord for some months already. So why don't I provide the binaries for PR https://github.com/ArduPilot/ArduRemoteID/pull/101 That should do the trick. That will be next week.  Another solution is to change PX4 implementation slightly, so it includes a timestamp in the empty OperatorID message. 

See https://github.com/ArduPilot/ArduRemoteID/blob/master/RemoteIDModule/transport.cpp#L63
This line gives the conditions when there is an issue with the Operator ID.

ryanjAA, 2024-01-26:
Notes re discord. Will build this firmware and give it a try but will try your binaries too. 

Omitting that line should do the trick. 

PX4 side will work to send a timestamp only and ideally localization (so if FAA do that, if EU send xxx). Will pick this up today and come back with findings.

ryanjAA, 2024-01-27:
@BluemarkInnovations is the bluemark fw signed? Or more specifically is lock level >0? I can’t upload newly built ota firmwares.

BluemarkInnovations, 2024-01-28:
@ryanjAA See https://github.com/ArduPilot/ArduRemoteID?tab=readme-ov-file#firmware-signing

So basically it is always signed and only allows OTA updates that are signed by the ArduRemoteID dev team keys or from BlueMark. Easiest method is to generate a private signing key yourself. Add the public key to the parameter of the ArdurRemoteID device e.g. to PUBLIC_KEY4. Then sign the OTA firmware with your key, afterwards it will accept your OTA update. Another work around is to flash via UART.

ryanjAA, 2024-01-28:
Noted. Just used flash for the time being but have it working on px4. Will post px4 changes to px4 and QGC and branch this off for the time being. 

Also, see different responses on different phone apps. Will post that too but that is all on app side.